HueForge-Official/HueForge #22: Rectangular STL's loads to the right cutoff in the model preview window and the "Drop Image Here" text

MajorGamerGeek, 2023-08-09:
**Description:** Rectangular STL's loads to the right cutoff in the model preview window and the "Drop Image Here" text

**Repro Steps:**

1. Launch HueForge v0.6.2
2. Click File -> File Open -> STL and select a rectangular STL file and Click Open
3. Observe the model preview window

**Result:**
Rectangular STL's loads to the right cutoff in the model preview window and the "Drop Image Here" text
[image]

**Expected Result:**
STLs should be centered in the model preview window and not cut off in anyway. Clicking Reset View does not fix the issue.

slavedas, 2023-08-09:
> Assigned #22 <https://github.com/HueForge-Official/HueForge/issues/22> to
> @slavedas <https://github.com/slavedas>.

The issue here is that the preview doesn't go away, not that the model
isn't centered in it's window

MajorGamerGeek, 2023-09-08:
Verified bug is fixed in HueForge 0.6.4. Closing bug as fixed.

closed 2023-09-08 by MajorGamerGeek