HueForge-Official/HueForge #31: c

MajorGamerGeek, 2023-09-11:
**Description:** Nothing occurs when the Add Filament menu options is selected with the Color Sliders.

**Repro Steps:**

1. Launch HueForge v0.6.4 RC2
2. Right click a Color Slider with no filament
3. Click on the Add Filament menu option

**Result:**
Nothing occurs when the Add Filament menu options is selected.

**Expected Results:**
Either the Add Filament button opened the New Filament Dialog or Remove the option entirely.

slavedas, 2023-09-11:
> Assigned #31 <https://github.com/HueForge-Official/HueForge/issues/31> to
> @slavedas <https://github.com/slavedas>.

It's worse than doing nothing.  It did something (added a filament) but
then didn't show the user it was added.  It is hidden in the list.

MajorGamerGeek, 2023-09-11:
Verified bug is fixed in HueForge v0.6.4. Closing bug as fixed.

closed 2023-09-11 by MajorGamerGeek