HueForge-Official/HueForge #37: Not all Color Sliders option display when you have less sliders then the opened project has configured

MajorGamerGeek, 2023-09-11:
**Description:** Not all Color Sliders option display when you have less sliders then the opened project has configured.

**Repro Steps:**

1. Launch HueForge v0.6.4 RC 3
2. Click the Remove a Slider - button until you have four Color Sliders
3. Click File -> Open File -> Project
4. Select the polymaker-stone-carving-nature-set and Click Open
5. Observe the HueForge preview window

**Result:**
Not all project configured Color Slider options display.
[image]

**Expected Results:**
Dynamically add the required number of Sliders needed for a opened HueForge Project.

MajorGamerGeek, 2023-12-10:
Bug still in build 0.7.0-Test release 12/09/23.

slavedas, 2023-12-11:
This should currently be fixed.

MajorGamerGeek, 2023-12-23:
Verified bug is fixed in HueForge version v0.7.0-Test 12/20/2023 build. Closing bug as fixed.

closed 2023-12-23 by MajorGamerGeek