HueForge-Official/HueForge #39: Unreleased PolyLite HF Filaments are in HueForge Filament Library and Blended Relief Cameo 0.08 Project

MajorGamerGeek, 2023-12-08:
**Bug Severity:** 3

**Description:** Unreleased PolyLite HF Filaments are in HueForge Filament Library and Blended Relief Cameo 0.08 Project

**Reproduction Steps:**
1.) Launch HueForge version v0.7.0-Test 12/15/2023
2.) Click on File -> Open File -> Project
3.) Select Blended Relief Cameo 0.08.hfp and Click Open
4.) Search for PolyLite HF within the Filament Library Filter

**Result:**
Four PolyLite HF Filaments appear in the Filament Library and two are used in the included project.

**Expected Result:**
The Four PolyLite HF Filament should not be in the shipping HueForge Filament Library and be in any included Projects.

slavedas, 2023-12-09:
The whole filament library needs to be worked on for this build so I'm going to use this issue as a marker to update the library.
Remove any personal filaments
Add Prusament that I have
Add updated IIID Max
Add Numakers
Add Overture
Add Paramount 3D
Add any new Protopasta (!)

MajorGamerGeek, 2023-12-10:
The PolyLite HF Filaments have been removed in HueForge 0.7.0 - Test  12/09/2023.

MajorGamerGeek, 2023-12-23:
I am working on printing a bunch of the test swatches for Protopasta HTPLA to expand their HueForge library.

MajorGamerGeek, 2024-01-27:
Bug still reproduce is fixed in HueForge v0.7.0-RC1. Unreleased PolyLite HF Filaments are in HueForge Filament Library and Blended Relief Cameo 0.08 Project.

MajorGamerGeek, 2024-01-27:
Bug fixed in HueForge v0.7.0-Beta1 with the unreleased PolyLite HF Filaments in the HueForge Filament Library and Blended Relief Cameo 0.08 Project.