HueForge-Official/HueForge #41: No Space between the Filament Color Name and Hex Code text within the These Filaments Are Updated Dialog

MajorGamerGeek, 2023-12-08:
**Bug Severity:** 3

**Description:** No Space between the Filament Color Name and Hex Code text within the These Filaments Are Updated Dialog

**Reproduction Steps:**
1.) Launch HueForge version v0.7.0-Test 12/07/2023
2.) Click the New Filament button within the Filament Library
3.) Click Ok on the Select Color Dialog
4.) Click Add on the Add Filament Dialog
5.) Click Save button within the Filament Library Dialog
6.) Observe the "These Filaments Are Updated. Would you like to save these changes?" dialog

**Result:**
No Space is displayed between the Filament Color Name and Hex Code text. (See Attached Screenshot)

[image]

**Expected Result:**
A space should be displayed between the Filament Color Name and Hex Code text

slavedas, 2023-12-08:
Fixed awaiting new build.

MajorGamerGeek, 2023-12-10:
Verified bug is fixed in HueForge build 0.7.0-Test date 12/09/23. Closing bug as fixed.

closed 2023-12-10 by MajorGamerGeek