HueForge-Official/HueForge #45: The "Drop Image Here" text is not centered in the preview image area

MajorGamerGeek, 2023-12-10:
**Bug Severity:** 3

**Description:** The "Drop Image Here" text is not centered in the preview image area

**Reproduction Steps:**
1.) Launch HueForge version v0.7.0-Test 12/09/2023
2.) Observe the UI

**Result:**
The "Drop Image Here" text is not centered in the preview image area. (See Attached Screenshot)

[image]

**Expected Result:**
The "Drop Image Here" text should be centered in the preview image area

slavedas, 2023-12-11:
This is fixed in the current test version, please confirm (image is centered as well)

MajorGamerGeek, 2023-12-23:
Verified bug is fixed in HueForge version v0.7.0-Test 12/20/2023 build. Verified both the "Drop Image Here" text and image you open in HueForge are correctly centered. Closing bug as fixed.

closed 2023-12-23 by MajorGamerGeek