HueForge-Official/HueForge #46: Copy Filament issues

slavedas, 2023-12-11:
Three issues I noticed - 1 when you click Copy filament, it should pop-up the dialog box immediately not just make a copy that you then have to find in the list.
Also when filtered and I clicked copy, both the original and copy vanished and turning off the filter didn't bring them back.

MajorGamerGeek, 2024-01-27:
Bug still reproduces in HueForge v0.7.0-RC1.

MajorGamerGeek, 2024-01-27:
Bug still reproduces in HueForge v0.7.0-Beta1.

MajorGamerGeek, 2024-02-08:
Bug still reproduces in HueForge v0.7.0-Beta3.

MajorGamerGeek, 2024-03-02:
Bug still reproduces in HueForge v0.7.1 - HotFix 3/1/24

MajorGamerGeek, 2024-03-12:
Bug still reproduces in HueForge v0.7.1 - HotFix 03/11/24

MajorGamerGeek, 2024-03-12:
The Filament Library filtering when copying a filament part of this bug is fixed in HueForge v0.7.1-Hotfix-03/12/24.

MajorGamerGeek, 2024-03-12:
Do we want to closing this bug as fixed and cut the first item listed in the bug of popping up a Filament Copy Dialog?

slavedas, 2024-03-12:
Yes, I think this is an acceptable solution to both.

closed 2024-03-12 by slavedas