HueForge-Official/HueForge #50: The Reset to Default Window Layout does not restore the hidden Lighting Settings and General Options and Operations windows

MajorGamerGeek, 2024-01-27:
**Description:** The Reset to Default Window Layout does not restore the hidden Lighting Settings and General Options and Operations windows.

**Repro Steps:**

1. Launch HueForge v0.7.0 RC1
2. Click Windows -> Lighting Settings to deselect the menu option
3. Click Windows -> General Options and Operations
4. Now click Windows -> Reset to Default Window Layout option
5. Observe the Reset HueForge UI

**Result:**
The Lighting Settings and General Options and Operations are not reset.

**Expected Results:**
All HueForge windows should be restored including the Lighting Settings and General Options and Operations windows.

MajorGamerGeek, 2024-01-27:
Bug still reproduces in HueForge v0.7.0-Beta1.

MajorGamerGeek, 2024-02-08:
Bug still reproduces in HueForge v0.7.0-beta-2-22-g584999d on Windows.

slavedas, 2024-02-08:
This is an odd one because it is relying on the layout from the first time
you launched and if your layout is corrupt, it won't work.

MajorGamerGeek, 2024-03-02:
Bug still reproduces in HueForge v0.7.1-Hotfix 3/1/24 on Windows.

MajorGamerGeek, 2024-03-12:
Bug still reproduces in HueForge v0.7.1-Hotfix 3/11/24 on Windows.

MajorGamerGeek, 2024-03-12:
Bug reproduces in HueForge v0.7.1-Hotfix 3/12/24 on Windows.