HueForge-Official/HueForge #61: Filaments from a users filament library and filaments from a project are displayed in a different font style within the Filament Library

MajorGamerGeek, 2024-01-30:
**Bug Severity:** 3

**Description:** 
Filaments from a users filament library and filaments from a project are displayed in a different font style within the Filament Library.

**Reproduction Steps:**
1.) Launch HueForge version v0.7.0-Beta-1
2.) File -> Open File -> Project
3.) Select Color_Pop_Cardinals.hfp and click
4.) Click on the filter edit box and search for Chocolate
5.) Observe the two Chocolate filaments text within the Filament Library

**Result:**
The found filaments are displayed in a different font style.
[image]

**Expected Result:**
Filaments from a users filament library and filaments from a project should be displayed in the same font style.

MajorGamerGeek, 2024-02-08:
Bug still reproduces in HueForge v0.7.0-beta-3 on both macOS and Windows.

slavedas, 2024-03-12:
I would fix this if I had any idea why it was doing it

slavedas, 2024-03-12:
It's true for Copy Filament new filaments as well though in my case I'm just seeing it being not bold where the normal font is bold

slavedas, 2024-03-12:
I take that back.  I believe I have solved it.

MajorGamerGeek, 2024-03-12:
Verified bug is fixed in HueForge v0.7.1-Hotfix 03/12/24. Closing bug as fixed.

closed 2024-03-12 by MajorGamerGeek