HueForge-Official/HueForge #77: Clicking the Add button within the Add Filament Dialog does not add a new Filament

MajorGamerGeek, 2024-03-02:
**Bug Severity:** 2

**Description:** 
Clicking the Add button within the Add Filament Dialog does not add a new Filament.

**Reproduction Steps:**
1.) Launch HueForge version v0.7.1-Hotfix 3/1/24
2.) Click the Add Filament Button under Filament Library
3.) Select any color and Click the Ok button
4.) Enter the Brand MyBrand and Name Test
5.) Finally Click Add button to add the new Filament 

**Result:**
Add Filament Dialog says "Enter new Filament Information" and your new filament is not added within your HueForge Filament Library.

**Expected Result:**
The users new Filament should be created within their Filament Library.

MajorGamerGeek, 2024-03-08:
Bug reproduces with the latest HueForge v0.7.1-03/07/24 build.

slavedas, 2024-03-08:
I have removed the Add button for the next build.  Add and Close works.

MajorGamerGeek, 2024-03-11:
Verified bug as Fixed in HueForge 0.7.1-Hotfix-3/10/24. Closing bug as fixed.

closed 2024-03-11 by MajorGamerGeek