HueForge-Official/HueForge #81: Create Filament -> Add Filament dialog defaults the filament type alwas as "IMAGE"

MajorGamerGeek, 2024-03-11:
**Bug Severity:** 3

**Description:**
Create Filament -> Add Filament dialog defaults the filament type "IMAGE"

**Reproduction Steps:**
1.) Launch HueForge version v0.7.1-Hotfix 3/10/24
2.) Click File -> Open File -> Image
3.) Select any image such as Color_Aware_Test.png and Click Open
4.) Right click anywhere on the project image and select Create Filament
5.) Observe New Filament Dialog

**Result:**
The default Filament type is inputted as "IMAGE"
[image]

**Expected Result:**
The currently selected Filament Library filament type should be automatically inputted into the Filament Type.

slavedas, 2024-03-11:
This was actually intentional. It's an image based Filament and it's likely
not real or in existence.

MajorGamerGeek, 2024-03-12:
Closing bug as by design in HueForge v0.7.1-Hotfix 03/11/24.

closed 2024-03-12 by MajorGamerGeek