HueForge-Official/HueForge #86: The currently opened project is being closed when a user selects Click File -> Open File -> Project

MajorGamerGeek, 2024-03-17:
**Bug Severity:** 2

**Description:** 
The currently opened project is being closed when a user selects Click File -> Open File -> Project

**Reproduction Steps:**
1.) Launch HueForge version v0.7.1-Hotfix 03/16/24 on Windows
2.) Click File -> Open File -> Project
3.) Select any project such as Blended Relief Cameo Example.hfp and click Open
5.) Now click File -> Open File -> Project
6.) Observe the HueForge UI

**Result:**
The File Load dialog opens and the currently open project is closed.

**Expected Result:**
The File Load dialog should open and the currently open project should stay opened.

slavedas, 2024-03-17:
This is definitely true. Expected behavior would be that the project stays open in case you cancel right?  Only close it if you actually load a new one.

MajorGamerGeek, 2024-03-17:
> This is definitely true. Expected behavior would be that the project stays open in case you cancel right? Only close it if you actually load a new one.

Exactly, the expected behavior would be that the project stays open in case a user cancels opening another project. It should only close the currently open project if you actually load another project.

MajorGamerGeek, 2024-03-24:
Verified bug is fixed in HueForge v0.7.1-03/23/24. Closing bug as fixed.

closed 2024-03-24 by MajorGamerGeek