HueForge-Official/HueForge #87: Color Pop luminance method projects are being rendered incorrectly in the HueForge mesh preview window

MajorGamerGeek, 2024-03-18:
**Bug Severity:** 2

**Description:**
Color Pop luminance method projects are being rendered incorrectly in the HueForge mesh preview window

**Reproduction Steps:**
1.) Launch HueForge version v0.7.1-Hotfix 03/16/24 on Windows
2.) Click File -> Open File -> Project
3.) Select any included Color Pop project such as "Blended Relief Cameo Example.hfp" or "Woman in Spacesuit Torso Color Pop.hfp" and click Open
5.) Observe the HueForge UI

**Result:**
The loaded color pop project is rendered incorrectly in mesh window:
[image]
[image]

**Expected Result:**
The File Load dialog should open and the currently open project should stay opened.

**Notes:**
Bug occurs with both Nvidia RTX 3060M and Intel Iris graphics.

slavedas, 2024-03-18:
I sent you two replacement projects.  Color Pop had multiple rendering issues that I believe I sorted and adjusted the projects for

MajorGamerGeek, 2024-03-24:
Verified bug is fixed in HueForge v0.7.1-Hotfix-03/23/24. Closing bug as fixed.

closed 2024-03-24 by MajorGamerGeek