HueForge-Official/HueForge #89: The Dragging and Dropping of images from online sources is not working in HueForge macOS

MajorGamerGeek, 2024-04-04:
**Bug Severity:** 2

**Description:** 
The dragging and dropping ofimage from online sources is not working in HueForge macOS

**Reproduction Steps:**
1.) Launch HueForge version v0.7.1-test-19-gcc04f2e on macOS Sonoma 14.4.1
2.) From Discord or a Web Browser attempt to drag and drop a support image type
3.) Observe the HueForge UI

**Result:**
Nothing occurs or the following error message displays:
[image]

**Expected Result:**
The Dragging and Drop of images from online sources should correctly open in HueForge v0.7.1 on macOS.

slavedas, 2024-04-04:
As I suspected, I was just missing a Sandbox permission.  I'll push a fixed version shortly.

MajorGamerGeek, 2024-04-04:
Verified bug is fixed in HueForge v0.7.1-test-20-g1321efd on macOS Sonoma 14.4.1. Closing bug as fixed.