HueForge-Official/HueForge #93: The Unable to Open File Dialog does not list *.svg as a supported file type

MajorGamerGeek, 2024-04-09:
Bug Severity: 3

Description:
The Unable to Open File Dialog does not list *.svg as a supported file type.

Reproduction Steps:
1.) Launch HueForge version v0.7.1-test-26-g132e83d on Windows
2.) Drag an unsupported file type onto the HueForge UI such as an *.exe
3.) Observe the Unable to Open File dialog

Result:
"File must be opened of the following: *.hfp, *.stl, *.jpg, *.jpeg, *.png, *.webp"
[image]

Expected Result:
The Unable to Open File Dialog should list *.svg as a supported file type.

slavedas, 2024-04-09:
> Assigned #93 <https://github.com/HueForge-Official/HueForge/issues/93> to
> @slavedas <https://github.com/slavedas>.

Also you can't open SVGs from any menu... good catch

MajorGamerGeek, 2024-04-09:
Verified bug is fixed in v0.7.1-beta-rc1 on macOS and Windows. Closing bug as fixed.

closed 2024-04-09 by MajorGamerGeek