HueForge-Official/HueForge #103: The Undock and Close buttons do not work on the Model Geometry window with a STL is loaded

MajorGamerGeek, 2024-07-20:
**Bug Severity:** 2

**Description:**
The Undock and Close buttons do not work on the Model Geometry window with a STL is loaded

**Reproduction Steps:**
1.) Launch HueForge version v0.7.3 
2.) Click File -> Open File -> STL
3.) Select any STL and Click the Open button
4.) Now click both the Close X button on the Model Geometry window to close
5.) Now click the Undock expand button on the Model Geometry window

**Result:**
Nothing occurs. The Model Geometry window stays docked and open.

**Expected Result:**
The Undock and Close buttons on the Model Geometry window should always work.

**Notes:**
Undocking the window and then clicking the red X works with a STL opened.

MajorGamerGeek, 2024-08-01:
Bug reproduces in HueForge v0.8.0-test-5.

MajorGamerGeek, 2024-08-04:
Bug reproduces in HueForge v0.7.3-test-rc1.

slavedas, 2024-08-04:
This is because the entire dock is disabled. Won't fix

MajorGamerGeek, 2024-08-05:
The Model Geometry dialog does not appear when opening STL's in previous v0.7.2 of HueForge. Closing bug as Won't Fix for HueForge v0.7.3.

closed 2024-08-05 by MajorGamerGeek