HueForge-Official/HueForge #115: HueForge does not center within the prediction windows and saved STL when a board is enabled

MajorGamerGeek, 2024-08-07:
**Bug Severity:** 1

**Description:**
HueForge does not center within the prediction windows and saved STL when a board is enabled

**Reproduction Steps:**
1.) Launch HueForge version v0.8.0-rc2 on Linux Mint 21.3 Cinnamon
2.) Now click File -> Open File -> Image
3.) Select any Image and Click Open
4.) Click the Border button within General Options and Operations
5.) Observe HueForge Prediction window

**Result:**
HueForge is not centered on the boarder within the prediction windows and saved STL 
[image]
[image]

**Expected Result:**
HueForge should be centered within the prediction window and it saved STL.

slavedas, 2024-08-07:
> to @slavedas <https://github.com/slavedas>.

Are you sure this isn't an issue with the source image not being centered
in the transparent region?

MajorGamerGeek, 2024-08-07:
> Are you sure this isn't an issue with the source image not being centered in the transparent region?
> […](#)

Yes I am sure I have loaded multiple include project images with and without transparent regions.f

MajorGamerGeek, 2024-08-07:
Bug reproduces on all platforms. 
[image]

MajorGamerGeek, 2024-08-08:
Verified bug is fixed in HueForge v0.8.0-beta-1. Closing bug as fixed.

closed 2024-08-08 by MajorGamerGeek