HueForge-Official/HueForge #116: Once a autosave project is created a user is prompted to restore it every time they launch HueForge

MajorGamerGeek, 2024-08-07:
**Bug Severity:** 3

**Description:**
Once a autosave project is created a user is prompted to restore it every time they launch HueForge

**Reproduction Steps:**
1.) Launch HueForge version v0.7.3-rc3 on Linux Mint 21.3 Cinnamon
2.) Now click File -> Open File -> Image
3.) Select any Image and Click Open
4.) Force close HueForge via task manger or other methods
5.) Launch HueForge again
6.) Click Ok on the "AutoSave Found" dialog to restore the autosaved project
7.) Save the project
8.) Close the project with File -> Close
9.) Now File -> Quit HueForge
10.) Launch HueForge

**Result:**
HueForge prompts again to restore the same autosave project.
[image]

**Expected Result:**
HueForge loads without prompting the user that an autosave project has been found since it has already been restored and saved.

slavedas, 2024-08-07:
> to @slavedas <https://github.com/slavedas>.

Yes, it only gets cleared out when you save a project. I guess I should
clear on close too?

MajorGamerGeek, 2024-08-07:
IMO: Clear after being restored and saved, being restored and closed gracefully or after being Canceled and Closed gracefully.

MajorGamerGeek, 2024-08-14:
Verified bug is fixed in both HueForge v0.7.3 and v0.8.0-test-1. Closing bug as fixed.

closed 2024-08-14 by MajorGamerGeek