HueForge-Official/HueForge #131: The Color Core and Mesh Core labels are being displayed under their own core

MajorGamerGeek, 2024-11-18:
**Bug Severity:** 3

**Description:**
The Color Core and Mesh Core labels are being displayed under their own core

**Reproduction Steps:**
1.) Launch HueForge version V0.8.0-Test-1.1 on macOS
2.) Observe HueForge

**Result:**
Color Core and Mesh Core labels are being displayed under their own core blocks.
[image]

**Expected Result:**
Color Core and Mesh Core labels display clearly above their core.

MajorGamerGeek, 2024-11-18:
Bug reproduce in v0.8.0-Test-1with the Linux App Image.
[image]

slavedas, 2024-11-18:
Fixed in windows for some projects, will test in other versions to confirm