HueForge-Official/HueForge #135: The Reset to Default Window Layout Menu Option does not reset Windows to their Default Layout

MajorGamerGeek, 2024-11-18:
**Bug Severity:** 2

**Description:**
The Reset to Default Window Layout Menu Option does not reset Windows to their Default Layout

**Reproduction Steps:**
1.) Launch HueForge version v0.8.0-Test-1 on macOS or Linux
2.) Click Windows and uncheck Color Silders, Filament Library, Lighting Settings, Model Geometry, and General Options and Operations
3.) Now Click Windows -> Reset To Default Window Layout menu option
4.) Observe HueForge

**Result:**
The Windows -> Reset to Default Window Layout Option does nothing.
[image]

**Expected Result:**
All Windows should be reset to their default Window Layout.

MajorGamerGeek, 2024-11-22:
Bug still reproduces in HueForge v0.8.1-rc-1.

slavedas, 2024-11-22:
Yes I fixed it locally but didn't update the binary last night

MajorGamerGeek, 2024-11-22:
Verified bug is fixed in HueForge v0.8.1-rc-1-1 on Windows.