HueForge-Official/HueForge #137: The Import Existing Configuration windows does not open when selected from the File Menu

MajorGamerGeek, 2024-11-18:
**Bug Severity:** 2

**Description:**
The Import Existing Configuration window does not open when selected from the File Menu

**Reproduction Steps:**
1.) Launch HueForge version v0.8.0-Test-1 on Windows
2.) Click File -> Import Existing Configuration
3.) Observe HueForge

**Result:**
Users is prompted with two Duplicate File Found dialog
[image]

**Expected Result:**
The Existing Configuration window displays with options on which configuration to import.

slavedas, 2024-11-18:
> to @slavedas <https://github.com/slavedas>.

This was on a brand new system?  No previous HueForge?

MajorGamerGeek, 2024-11-18:
> o

This was not a brand new system.

slavedas, 2024-11-18:
Yeah I figured out this error.  I had accidentally deleted a conditional so
it was ALWAYS running.

MajorGamerGeek, 2024-11-22:
Verified bug is fixed in HueForge v0.8.1-rc1. Closing bug as fixed.

closed 2024-11-22 by MajorGamerGeek