HueForge-Official/HueForge #141: HueForge project images and prediction windows display small and shifted left when opened via the taskbar

MajorGamerGeek, 2024-11-22:
**Bug Severity:** 2

**Description:**
HueForge project images and prediction windows display small and shifted left when opened via the taskbar

**Reproduction Steps:**
1.) Launch HueForge version v0.8.1-rc1
2.) Right click on the HueForge icon on the taskbar and select any recent project
3.) Observe the HueForge UI

**Result:**
The opened project image and prediction windows open very small and shifted left.
[image]

**Expected Result:**
The opened project image and predictions display in their normal location and size.

MajorGamerGeek, 2024-11-22:
Yes this was found on Windows.

slavedas, 2024-11-22:
Cannot reproduce