HueForge-Official/HueForge #144: Cannot open any STL file

advancingu, 2024-12-07:
When trying to open an STL file with `File -> Open File -> STL` in the top menu and selecting a .stl file and then pressing `Open`, the following error message is shown:

[image]

I tried opening from other directories including `/tmp`, files without filenames or any potentially difficult characters, various `.stl` files but none can be opened.

HueForge version is 0.8.1, launched from AppImage under vanilla Ubuntu 24.04.1

bonn2, 2025-02-10:
I am having the same problem, in my testing it appears to have appeared in the 0.8.0 release as running 0.7.4 allows me to open stls just fine.

I am also running the AppImage but under Manjaro Stable 6.12.1-1

slavedas, 2025-02-16:
> to @slavedas <https://github.com/slavedas>.

Huh?  I loaded an STL on steam last night...

MajorGamerGeek, 2025-03-31:
Bug is still reproduce with larger STL files on Linux Mint 21.3 Cinnamon with MESA 23.2.1.

MajorGamerGeek, 2025-03-31:
Some STL files do load successfully on Linux v0.9.0-Test-4.1 AppImage. Opened new bug for crashing bug.