HueForge-Official/HueForge #150: HueForge crashes to desktop when coping a core selection from the Color Core to the Mesh Core in a Color Match project

MajorGamerGeek, 2025-01-31:
**Bug Severity:** 1

**Description:**
HueForge crashes to desktop when coping a core selection from the Color Core to the Mesh Core in a Color Match project

**Reproduction Steps:**
1.) Launch HueForge version v0.8.2-Test on Windows
2.) Click File -> Open File -> Project
3.) Select any Color Match Project such as Frog Color Match Example.hfp and click Open
4.) Highlight any section within the Color Core 
5.) Right Click on the Color Core and then select Copy Section
6.) Right click on the Mesh Core and then select Paste Section insert or Paste Section Replace

**Results:**
The HueForge crash to the desktop with the following Access Violation:
Exception code: 0xc0000005
Exception: Access Violation
0x7ffdbfcb1521: memmove - 140727821210112

[lv.log](https://github.com/user-attachments/files/18615474/lv.log)
[crash_file.txt](https://github.com/user-attachments/files/18615475/crash_file.txt)

**Expected Result:**
The HueForge pastes the core selection into the Mesh Core and continues running as normal.

slavedas, 2025-01-31:
> to @slavedas <https://github.com/slavedas>.

Should be fixed now.  Was an inline function vs static issue I believe.

MajorGamerGeek, 2025-02-02:
Verified bug is fixed in the latest Windows v0.8.2-Test build. Closing bug as fixed.

closed 2025-02-02 by MajorGamerGeek