HueForge-Official/HueForge #153: The brightness compensation graph is incorrectly being displayed in the top right of the prediction area

MajorGamerGeek, 2025-03-23:
**Bug Severity:** 3

**Description:**
The brightness compensation graph is incorrectly being displayed in the top right of the prediction area

**Reproduction Steps:**
1.) Launch HueForge version v0.9.0-Test-3.1 on macOSX
2.) Observe the HueForge prediction window

**Results:**
The brightness compensation graph is displayed in the HueForge prediction area

[image]

**Expected Result:**
The brightness compensation graph should be displayed within the top right of the HueForge source image area.

slavedas, 2025-03-23:
> to @slavedas <https://github.com/slavedas>.

This is correct behavior when not in Edit Image mode

MajorGamerGeek, 2025-03-23:
Closing as by design.

closed 2025-03-23 by MajorGamerGeek