HueForge-Official/HueForge #159: HueForge crashes when loading large STL files on Linux

MajorGamerGeek, 2025-03-31:
**Bug Severity:** 1

**Description:**
HueForge crashes when loading large STL files on Linux

**Reproduction Steps:**
1.) Launch HueForge version v0.9.0-Test-4.1 AppImage on Linux
2.) Select File -> Open File -> STL
3.) Select an Large STL file 100mb or greater
4.) Finally click Open

**Results:**
HueForge crashes to desktop. (See attached crash logs)

[crash_file.txt](https://github.com/user-attachments/files/19533579/crash_file.txt)
[lv.log](https://github.com/user-attachments/files/19533580/lv.log)

**Expected Result:**
The large STL file loads successfully int HueForge.

slavedas, 2025-03-31:
> to @slavedas <https://github.com/slavedas>.

I am unable to reproduce this on my Linux Box.  STLs well beyond the Max
Triangles limit run fine - loaded 198mb.  HOWEVER if I try to grab sliders
on the Color Core, they crashed and I fixed that issue.

slavedas, 2025-03-31:
I confess I didn't try File->Open File ->STL I did drag and drop, but it should be the same

MajorGamerGeek, 2025-03-31:
I updated the reproduction steps to be more reliable crashing.