HueForge-Official/HueForge #160: HueForge crashes when disabling Full Range while all sliders at 0 or disabled

MajorGamerGeek, 2025-03-31:
Bug Severity: 1

Description:
HueForge crashes when disabling Full Range while all sliders at 0 or disabled

Reproduction Steps:
1.) Launch HueForge version v0.9.0-Test-4.1 on any platform
2.) Disable all sliders above 0
3.) Click the Full Range button

Results:
HueForge crashes to desktop. (See attached crash logs)

[crashlog.txt_2025-03-31_10-27-46.txt](https://github.com/user-attachments/files/19533837/crashlog.txt_2025-03-31_10-27-46.txt)
[lv.log](https://github.com/user-attachments/files/19533838/lv.log)

Expected Result:
Full Range disables and HueForge continues to run normally

slavedas, 2025-03-31:
Fixed in Linux-test-5 and being merged to other versions

MajorGamerGeek, 2025-04-06:
Verified bug is fixed in HueForge v0.9.0-Beta-1 on macOS and Windows builds. Closing bug as fixed.

closed 2025-04-06 by MajorGamerGeek